adafruit/TinyLoRa #9: problems compiling for feather M0 Express

jerryneedell, 2018-10-21:
@brentru  I have gotten this to work with a Feather M0 Express, but i have had to make a few changes.
Theseare due to compiler differences between the AVR and ARM compilers
In TinyLoRa.cpp
Replace "static const" with "const"   -- I'm not sure this was the only way to make it work, but it would not compile with "static const" 

In TinyLora.h:
remove include for delay  -- it is not needed and /util/delay does not exist -  delay is available without this include

select SF12BW125  for the datarate

Here are the "diffs"
```
diff --git a/TinyLoRa.cpp b/TinyLoRa.cpp
index 4647985..79b90ac 100644
--- a/TinyLoRa.cpp
+++ b/TinyLoRa.cpp
@@ -30,7 +30,7 @@ static SPISettings RFM_spisettings = SPISettings(4000000, MSBFIRST, SPI_MODE0);
 */
 
 #ifdef AU915
-static const unsigned char PROGMEM TinyLoRa::LoRa_Frequency[8][3] = {
+const unsigned char PROGMEM TinyLoRa::LoRa_Frequency[8][3] = {
        { 0xE5, 0x33, 0x5A },   //Channel 0 916.800 MHz / 61.035 Hz = 15020890 = 0xE5335A
        { 0xE5, 0x40, 0x26 },   //Channel 2 917.000 MHz / 61.035 Hz = 15024166 = 0xE54026
        { 0xE5, 0x4C, 0xF3 },   //Channel 3 917.200 MHz / 61.035 Hz = 15027443 = 0xE54CF3
@@ -43,7 +43,7 @@ static const unsigned char PROGMEM TinyLoRa::LoRa_Frequency[8][3] = {
 #endif
 
 #ifdef EU863
-static const unsigned char PROGMEM TinyLoRa::LoRa_Frequency[8][3] = {
+const unsigned char PROGMEM TinyLoRa::LoRa_Frequency[8][3] = {
        { 0xD9, 0x06, 0x8B },   //Channel 0 868.100 MHz / 61.035 Hz = 14222987 = 0xD9068B
        { 0xD9, 0x13, 0x58 },   //Channel 1 868.300 MHz / 61.035 Hz = 14226264 = 0xD91358
        { 0xD9, 0x20, 0x24 },   //Channel 2 868.500 MHz / 61.035 Hz = 14229540 = 0xD92024
@@ -57,7 +57,7 @@ static const unsigned char PROGMEM TinyLoRa::LoRa_Frequency[8][3] = {
 #endif
 
 #ifdef US902
-static const unsigned char PROGMEM TinyLoRa::LoRa_Frequency[8][3] = {
+const unsigned char PROGMEM TinyLoRa::LoRa_Frequency[8][3] = {
        { 0xE1, 0xF9, 0xC0 },           //Channel 0 903.900 MHz / 61.035 Hz = 14809536 = 0xE1F9C0
        { 0xE2, 0x06, 0x8C },           //Channel 1 904.100 MHz / 61.035 Hz = 14812812 = 0xE2068C
        { 0xE2, 0x13, 0x59},            //Channel 2 904.300 MHz / 61.035 Hz = 14816089 = 0xE21359
@@ -70,7 +70,7 @@ static const unsigned char PROGMEM TinyLoRa::LoRa_Frequency[8][3] = {
 #endif
 
 #ifdef AS920
-static const unsigned char PROGMEM TinyLoRa::LoRa_Frequency[8][3] = {
+const unsigned char PROGMEM TinyLoRa::LoRa_Frequency[8][3] = {
        { 0xE6, 0xCC, 0xF4 },           //Channel 0 868.100 MHz / 61.035 Hz = 15125748 = 0xE6CCF4
        { 0xE6, 0xD9, 0xC0 },           //Channel 1 868.300 MHz / 61.035 Hz = 15129024 = 0xE6D9C0
        { 0xE6, 0x8C, 0xF3 },           //Channel 2 868.500 MHz / 61.035 Hz = 15109363 = 0xE68CF3
@@ -89,7 +89,7 @@ static const unsigned char PROGMEM TinyLoRa::LoRa_Frequency[8][3] = {
 *****************************************************************************************
 */
 
-static const unsigned char PROGMEM TinyLoRa::S_Table[16][16] = {
+const unsigned char PROGMEM TinyLoRa::S_Table[16][16] = {
          {0x63,0x7C,0x77,0x7B,0xF2,0x6B,0x6F,0xC5,0x30,0x01,0x67,0x2B,0xFE,0xD7,0xAB,0x76},
          {0xCA,0x82,0xC9,0x7D,0xFA,0x59,0x47,0xF0,0xAD,0xD4,0xA2,0xAF,0x9C,0xA4,0x72,0xC0},
          {0xB7,0xFD,0x93,0x26,0x36,0x3F,0xF7,0xCC,0x34,0xA5,0xE5,0xF1,0x71,0xD8,0x31,0x15},
:...skipping...
diff --git a/TinyLoRa.cpp b/TinyLoRa.cpp
index 4647985..79b90ac 100644
--- a/TinyLoRa.cpp
+++ b/TinyLoRa.cpp
@@ -30,7 +30,7 @@ static SPISettings RFM_spisettings = SPISettings(4000000, MSBFIRST, SPI_MODE0);
 */
 
 #ifdef AU915
-static const unsigned char PROGMEM TinyLoRa::LoRa_Frequency[8][3] = {
+const unsigned char PROGMEM TinyLoRa::LoRa_Frequency[8][3] = {
        { 0xE5, 0x33, 0x5A },   //Channel 0 916.800 MHz / 61.035 Hz = 15020890 = 0xE5335A
        { 0xE5, 0x40, 0x26 },   //Channel 2 917.000 MHz / 61.035 Hz = 15024166 = 0xE54026
        { 0xE5, 0x4C, 0xF3 },   //Channel 3 917.200 MHz / 61.035 Hz = 15027443 = 0xE54CF3
@@ -43,7 +43,7 @@ static const unsigned char PROGMEM TinyLoRa::LoRa_Frequency[8][3] = {
 #endif
 
 #ifdef EU863
-static const unsigned char PROGMEM TinyLoRa::LoRa_Frequency[8][3] = {
+const unsigned char PROGMEM TinyLoRa::LoRa_Frequency[8][3] = {
        { 0xD9, 0x06, 0x8B },   //Channel 0 868.100 MHz / 61.035 Hz = 14222987 = 0xD9068B
        { 0xD9, 0x13, 0x58 },   //Channel 1 868.300 MHz / 61.035 Hz = 14226264 = 0xD91358
        { 0xD9, 0x20, 0x24 },   //Channel 2 868.500 MHz / 61.035 Hz = 14229540 = 0xD92024
@@ -57,7 +57,7 @@ static const unsigned char PROGMEM TinyLoRa::LoRa_Frequency[8][3] = {
 #endif
 
 #ifdef US902
-static const unsigned char PROGMEM TinyLoRa::LoRa_Frequency[8][3] = {
+const unsigned char PROGMEM TinyLoRa::LoRa_Frequency[8][3] = {
        { 0xE1, 0xF9, 0xC0 },           //Channel 0 903.900 MHz / 61.035 Hz = 14809536 = 0xE1F9C0
        { 0xE2, 0x06, 0x8C },           //Channel 1 904.100 MHz / 61.035 Hz = 14812812 = 0xE2068C
        { 0xE2, 0x13, 0x59},            //Channel 2 904.300 MHz / 61.035 Hz = 14816089 = 0xE21359
@@ -70,7 +70,7 @@ static const unsigned char PROGMEM TinyLoRa::LoRa_Frequency[8][3] = {
 #endif
 
 #ifdef AS920
-static const unsigned char PROGMEM TinyLoRa::LoRa_Frequency[8][3] = {
+const unsigned char PROGMEM TinyLoRa::LoRa_Frequency[8][3] = {
        { 0xE6, 0xCC, 0xF4 },           //Channel 0 868.100 MHz / 61.035 Hz = 15125748 = 0xE6CCF4
        { 0xE6, 0xD9, 0xC0 },           //Channel 1 868.300 MHz / 61.035 Hz = 15129024 = 0xE6D9C0
        { 0xE6, 0x8C, 0xF3 },           //Channel 2 868.500 MHz / 61.035 Hz = 15109363 = 0xE68CF3
@@ -89,7 +89,7 @@ static const unsigned char PROGMEM TinyLoRa::LoRa_Frequency[8][3] = {
 *****************************************************************************************
 */
 
-static const unsigned char PROGMEM TinyLoRa::S_Table[16][16] = {
+const unsigned char PROGMEM TinyLoRa::S_Table[16][16] = {
          {0x63,0x7C,0x77,0x7B,0xF2,0x6B,0x6F,0xC5,0x30,0x01,0x67,0x2B,0xFE,0xD7,0xAB,0x76},
          {0xCA,0x82,0xC9,0x7D,0xFA,0x59,0x47,0xF0,0xAD,0xD4,0xA2,0xAF,0x9C,0xA4,0x72,0xC0},
          {0xB7,0xFD,0x93,0x26,0x36,0x3F,0xF7,0xCC,0x34,0xA5,0xE5,0xF1,0x71,0xD8,0x31,0x15},
@@ -972,4 +972,4 @@ void TinyLoRa::AES_Calculate_Round_Key(unsigned char Round, unsigned char *Round
       Temp[j]                   = Round_Key[j + (i << 2)];
     }
   }
-}
\ No newline at end of file
+}
diff --git a/TinyLoRa.h b/TinyLoRa.h
index 4250aae..5519f6b 100644
--- a/TinyLoRa.h
+++ b/TinyLoRa.h
@@ -2,7 +2,7 @@
 #define TINY_LORA_H
 
 #include <Arduino.h>
-#include <util/delay.h>
+//#include <util/delay.h>
 #include <avr/pgmspace.h>
 
 // debugging, unset if you don't need this.
@@ -19,12 +19,12 @@
 //#define AS920
 // Define fixed datarate
 //#define SF7BW125
-//#define SF12BW125
+#define SF12BW125
 //#define SF11BW125
 //#define SF10BW125
 //#define SF9BW125
 //#define SF8BW125
-#define SF7BW250
+//#define SF7BW250
 
 /* RFM Modes */
 #define MODE_SLEEP  0x00
(END)
```

another issue I ran into was that, in the function senddata 
 https://github.com/adafruit/TinyLoRa/blob/master/TinyLoRa.cpp#L342
the call to EncryptPayload  overwrites the contents of Data with the encrypted version.
Since the example program only initializes the Data array at the start, it is encrypted and sent properly the first time it is called, but then it encrypts the encrypted data .... the messages sent after the first one are  not readable when received.
In normal use, I don't think this matters since the data array will usually be repopulated with each message.

To verify this, made these changes to the example:
instead of a string use array of bytes initialized to {0,1,2,3,4,5,6,7,8,9,10) althought this is never sent
reconstruct the array Data just before sending it -- I populate it with {40,41,42,43,44,45,46,47,48,49,50} since they are printable ASCII characters.
By repopulating the array each time it works for me.

 
```

jerryneedell@Ubuntu-Macmini:~/Arduino/hello_LoRa_gjn$ diff hello_LoRa_gjn.ino ../libraries/TinyLora/examples/hello_LoRa/hello_LoRa.ino 

< unsigned char loraData[11] = {0,1,2,3,4,5,6,7,8,9,10};
---
> unsigned char loraData[11] = {"hello LoRa"};
32c32
< TinyLoRa lora = TinyLoRa(11, 12);
---
> TinyLoRa lora = TinyLoRa(7, 8);
37,38c37,38
< //  while (! Serial);
<   Serial.begin(115200);
---
>   while (! Serial);
>   Serial.begin(9600);
45d44
<   SPI.begin();
51d49
<   uint16_t i;
53,56d50
<   for(i=0;i<11;i++)
<   {
<     loraData[i]=64+i;
<   }
```

Let me know if you need any more information or clarification.

jerryneedell, 2018-10-21:
hmm -- odd stuff happening -- I am now finding that the code from this repo does not work reliably - sometimes it does - sometime not. My gateway does not see any traffic from it sometimes.
If I revert to @brentru repo https://github.com/brentru/TinyLoRa  then it works reliably. Clearly I'v missed something.... I'll keep poking and prodding it.
I make that same compiler related changes to both libraries and am using the data rate setting:
SF7BW125 on both.

brentru, 2018-10-21:
@jerryneedell  Did you disable the frame counter on the M0? Is your code set up to broadcast over a single or multi channel? If it's set to broadcast on single channel and you have a multi-channel gateway, the gateway's radios may not be listening on that frequency and miss it.

jerryneedell, 2018-10-22:
This may be a repeated message - I posted it, but don't see it so I'll try again
I think there is a typo in the adafruit repo in TinyLora.cpp in this line.
https://github.com/adafruit/TinyLoRa/blob/master/TinyLoRa.cpp#L207
the #define FULLCH should be #define MULTICH
I made this change and it is working reliably for me now with the adafruit repo.

I do have the frame counter disabled,

jerryneedell, 2018-10-22:
Inaddition to removing the "static" from "static const" For the Feather_M0  all uses of PROGMEM can be removed in TinyLora.cpp and the #include <avr/pgmspace.h> can be removed from TinyLora.h.  Perhaps these differences can be accommodated via "#ifdef"

jerryneedell, 2018-10-22:
@brentru I applied PR#4 and now the only changes I have to make are for the compiler differences. I see that you fixed the MULTICH definition there. Looks good!
Note: I also noted that you took the pin definitions out of the calling arguments for TinyLora() and they are set in TinyLora.h -- I also changed them for my configurations

brentru, 2018-10-23:
Thank you for the testing, @jerryneedell. I've modified `master` to reflect the channel selection and SF modifications you mentioned.

jerryneedell, 2018-10-23:
pulled current master -- made the usual compiler required changes and it works fine for me. Thanks!

jerryneedell, 2018-10-24:
I've narrowed this down and the only really necessary change for the M0 compiler is to replace all the 
"static const" with "const".
With this change, the sketches compile for both the 32U4 and the M0/

brentru, 2018-11-06:
@jerryneedell  Thank you for your continued testing - I'm going to add support for this in the next PR

brentru, 2018-11-06:
Added to master, closing...

closed 2018-11-06 by brentru