adafruit/TinyLoRa #11: modify examples to avoid corrupting fixed message

jerryneedell, 2018-11-07:
In the examples, a fixed string is created to send the message "Hello LoRa"
but when the message to be transmitted is processed, the contents of the data array passed to senData are destroyed during the encryption  process.

If the message is only initialized at  declaration then it is sent properly the first time, then garbled every time after that as the array is repeatedly encrypted.

It should be made clear that sendData destroys the data array.
The example can be easily fixed with a few simple changes:

replace the declaration
```
unsigned char loraData[11] = {"hello LoRa"};
```
with
```
char loraData[11];
---

in the loop
replace
```
lora.sendData(loraData, sizeof(loraData), lora.frameCounter);
```
with
```
strcpy(loraData,"Hello Lora");
lora.sendData((unsigned char *)loraData, sizeof(loraData), lora.frameCounter);
```

brentru, 2018-11-07:
> There are many ways to deal with this. We could have sendData make a local copy of the message, but that is a potential waste of up to 252 bytes.

Thank you for reminding me, I remember we spoke about this issue a few weeks back. I opened a new branch for this issue (https://github.com/adafruit/TinyLoRa/tree/fix-encrypt) which copies over the packet to in lora.sendData(*data*) to a new character array which is fed as an argument to the `Encrypt_Payload` function and doesn't modify the original value sent. 

Tested on my Feather 32u4 and TTN and received the same payload back every frame, instead of garbled like before.

brentru, 2018-11-07:
A patch for this issue has been merged into master. Closing...

closed 2018-11-07 by brentru