adafruit/TinyLoRa #12: information on disabling Frame Counter Check

jerryneedell, 2018-11-07:
This is not really an issue, but something that was puzzling me so I wanted to look into it a bit.
Feel free to close this at any time.

In setting up the devices on The Things Network (TTN) it was recommended that  Frame Counter Checks be disabled to avoid some cases where data reception stopped. 

Looking into this a bit, that makes sense.
See
https://www.thethingsnetwork.org/docs/lorawan/security.html#frame-counters

For simple examples, I suspect the security impact is not all that important, but I think this needs to be understood by users.  

If I read this correctly, in order to use the Frame Counter Check, one would have to re-register the device every time it is reset or if the frame counter overflows. It's not quite that bad, you could reset the frame counter to a value larger than the last transmission and it whorl be be OK to restart. Certainly not something to do during development! I think it is OK to disable it but some explanation should be included in the guide. When you disable it TTN gives a cryptic warning about how this should only be done during development and greatly diminishes security. Unfortunately, TTN does not explain what at that time.

brentru, 2018-11-07:
Noted - adding this into the guide/documentation. Thank you for providing extra information about the security implications of disabling frame counter check.

brentru, 2018-11-07:
Added to guide, closing...

closed 2018-11-07 by brentru