adafruit/TinyLoRa #18: Starting LoRa...Failed

flemssound, 2019-06-04:
I have bought a Adafruit Feather M0 Lora.

When I now try to send data via the hello_lora example sketch it will always Fail.
I have commentet out the right board in the configuration.

`
// Data Packet to Send to TTN
unsigned char loraData[11] = {"hello LoRa"};

// How many times data transfer should occur, in seconds
const unsigned int sendInterval = 5;

// Pinout for Feather 32u4 LoRa
//TinyLoRa lora = TinyLoRa(7, 8);

// Pinout for Feather M0 LoRa
TinyLoRa lora = TinyLoRa(3, 8);
`

brentru, 2019-06-04:
Does it fail to compile, or fail to send data? Could you post the output from your serial monitor?

brentru, 2019-06-04:
@flemssound Ok - I just attempted this on a Feather M0 Radio LoRa with the same pinout and could not replicate. Are you positive Pin 6 is connected to IO#1?

flemssound, 2019-06-05:
Compiling dosn't fail.
I have tried sending data again. And now it is working perfectly also without connecting Pin 6 and IO#1.
I have no clue why it is working now.

closed 2019-06-05 by brentru