adafruit/TinyLoRa #23: TinyLora with Feather Mo LoRa - Starting LoRa...Failed

wagroves223, 2019-07-08:
I am trying to get the tinylora_dht22 example to work with a Feather M0 LoRa but get 

"Starting LoRa...Failed" 
"Check your radio" 

I used this line of code for pinouts

// Pinout for Adafruit Feather M0 LoRa
TinyLoRa lora = TinyLoRa(3, 8);

I do have a jumper between IO1 and pin 6. Nothing else is connected currently.

Not sure what else to try - this same unit does successfully connect to TTN using the LMIC OTAA DHT22 example.

lloydXmas, 2019-07-18:
@wagroves223 I am also experiencing this. It may be an issue with the HopeRF module: it seems that reading from address 0x42 doesn't return a consistent RegVersion. 
Try commenting-out the lines 305-307 in TinyLoRa.cpp to skip the `RegVersion` check:  
https://github.com/adafruit/TinyLoRa/blob/436b01738dc353aad5287062cd9edd848f1a4b94/TinyLoRa.cpp#L304-L307

Like so:
``` cpp
  uint8_t ver = RFM_Read(0x42);
  // if(ver!=18){
  //   return 0;
  // }
```

wagroves223, 2019-07-18:
That solved the problem - thank you!

brentru, 2019-07-18:
@lloydXmas  @wagroves223  Thanks for the tip! Possibly HopeRF is shipping with different version numbers. One of the datasheets I have lists `0x12` and another (from HopeRF directly) lists `0x11` as a possible `RegVersion` value.

Could you add the following lines underneath `uint8_t ver = RFM_READ(0x42);` and let me know the result?
```
Serial.print("Version : "); 
Serial.println(ver, HEX);
```

wagroves223, 2019-07-18:
Results for the two Feather M0 Lora boards that I have:

10:03:00.586 -> Starting LoRa...Version : 0
10:03:00.586 -> OK
10:03:00.586 -> Sending LoRa Data...
10:03:00.641 -> Frame Counter: 0
10:03:01.656 -> delaying... 

10:08:25.781 -> Starting LoRa...Version : 12
10:08:25.781 -> OK
10:08:25.781 -> Sending LoRa Data...
10:08:25.831 -> Frame Counter: 0
10:08:26.846 -> delaying...

brentru, 2019-07-18:
@wagroves223 The register returned from the first board is odd. Is there a jumper wire between the Feather's Digital Pin 6 and Feather Pin io1? Second result is what the library expects.

wagroves223, 2019-07-18:
Both boards run hello_lora with or without the jumper between Pin 6 and IO1 and return the version numbers shown above with or without the jumper.

brentru, 2019-07-18:
@wagroves223  Does the first board, which returns version `0x00`, connect and send data to your gateway after you comment out the version check?

wagroves223, 2019-07-18:
Yes.  I have been using the same version of the sketch for both boards so the ABP parameters were not changed - there is only one device currently registered in the TTN application.  I was about to register the second device and try to run both at the same time.

wagroves223, 2019-07-18:
OK both nodes connecting and sending data using sketches with version check commented out.

[image]

brentru, 2019-07-19:
@wagroves223 Both boards I have return 0x12 and not 0x00. I spent some time this morning trying to replicate, it _sometimes_ returns 0x00 as the version but seems to return 0x12 on the next upload.

Thanks for helping me narrow this down more... You mentioned in the forums that you had success with the MCCI LMIC library. Was this with the board which returns a version of 0x00, or the other board?

One of the differences between the two libraries is the LMIC library toggles the RFM9x's RST  before communicating with the chip. I can add this into TinyLoRa and create a branch for you to try it out, if that's the case.

wagroves223, 2019-07-20:
Yes both Feather M0 LoRa boards (including the one that returned version 0x00) worked as is with the ttn-otaa-feather-us915-dht22 sketch.  I can try out a revised version of TinyLoRa.

brentru, 2019-07-22:
@wagroves223  that'd be great, the version I just released (2.0.0) is in the library manager if you'd like to update to it.

wagroves223, 2019-07-22:
Using the new version, I am back to the one board (0x00) failing to initialize.  The second board does work with the new helloLoRa sketch (0x12).  Commenting out the version check gets the first board to run the sketch.

brentru, 2019-07-22:
@wagroves223 I'm unable to reproduce this issue with the new version of TinyLoRa running on my board. Could you email support@adafruit.com and tell them you'd like to return the product with RMA "FORPHIL"? They can send you a replacement Feather M0 LoRa and we'll take this one for testing.

Thanks!

wagroves223, 2019-07-24:
Thank you for the assistance.  I’m in the middle of a project and now that two of the feather m0 lora boards are working (albeit with the modified code) I’m going to keep moving forward.  I have a few more of these boards – some with the sd card and the lora on a wing that I might eventually get back to trying with the tinylora library to see if there are any other RFM95 version issue problems.

Will be taking the two feather m0 lora boards into the field for a dust sensor project sometime in August and am looking forward to characterizing range etc.  Thanks again for the help.

WG

brentru, 2019-07-24:
Ok, good luck with the project!

closed 2019-07-24 by brentru

nyfelix, 2020-04-29:
This bug seem still to be present.
Today I pulled the latest version (1.2.2)  of TinyLora from plattform.io registry and my Feather M0 Lora with HopeRF Model RFM95C LoraChip did not start.
After commenting out the version check il worked well.

darrylhodgins, 2020-10-02:
@brentru, I'm seeing the same behaviour with my shiny new Feather M0 LoRa board and TinyLora library v. `1.4.0` (with the only change being that I've added the `Version : <number>` debugging output).  I've got:

```
// IRQ = 3
// NSS = 8
// RST = 4
// Pin 6 jumper to DIO1
TinyLoRa lora = TinyLoRa(3, 8, 4);
```

…and the serial output:

```
Starting LoRa...Version : 0
Failed
Check your radio
```

However, I am able to get it to work intermittently by pressing the reset button a number of times, until it works:

```
Starting LoRa...Version : 12
OK
```

I can see you added the reset toggle in #24, so I've tried messing with the timing in `TinyLoRa::begin()`, but that seems to make no difference.  Do you have any ideas?

Thanks :)

brentru, 2020-10-02:
@darrylhodgins  Do you have a physical jumper btween IO1 and feather m0 pin 6?

darrylhodgins, 2020-10-02:
@brentru, indeed I do.

I also tested it with the MCCI LoRaWAN LMIC library (v. `3.2.0`), which also needs that jumper from `IO1` to `6`.  The MCCI library seems to work just fine (though I couldn't figure out how to run it in single-channel mode with a Pi bonnet single-channel gateway… but that's another thing).

brentru, 2020-10-02:
@darrylhodgins  Are you using a Feather RFM96 or RFM95?

darrylhodgins, 2020-10-02:
@brentru: I'm using the [Feather M0 with RFM95](https://www.adafruit.com/product/3178).

morrischen1, 2020-12-18:
Hello! I am currently trying to get version number but I've gotten many different results. The default setting RFM_Read(REG_VER); which gives out the version number 0. I've tested RFM_Read(0x42); which gives out either version number 0 or 92. Then I tested RFM_Read(0x5C); which gives out different version numbers: 0 , 5 , 48 , 108 and 247. The most common version number that was appearing after each reset was 48. Does this mean that the version number is 48 for 0x5C?

I am trying to figure out why my M0 LoRa won't continue after "Sending LoRa Data...". It was previously stuck on "Starting LoRa..."  until I made the follwing lines to comments:
//uint8_t ver = RFM_Read(REG_VER);
  //if (ver != RFM9x_VER)
    //return 0;

brentru, 2020-12-18:
@morrischen1  What hardware are you using?

morrischen1, 2020-12-18:
@brentru I am using Feather M0 RFM69HCW and I have connected IO1 to pin 6 during my testing with different 0xXX

morrischen1, 2020-12-18:
@brentru sorry for taking your time, it turns out i have been given the wrong board.

morrischen1, 2020-12-19:
@brentru I just noticed I have gotten the wrong board (Feather M0 RFM69HCW) but it shouldn't be a problem. I and my friend took the hello_lora example code and modified it a bit so it would send a fixed value. We have the exact same code but once we tried our boards on our own computer we got different results. My friend who has a Feather M0 RFM9x LoRa Radio got to the printing of framecounter and "delaying" while I was stuck on "Starting LoRa". In the code, we also wrote our own NwkSkey etc. After the result, I took his board and tested his board on my code which is the same code with different keys and got the result as before "Starting LoRa". Then he plugged in my board to his computer with his code and his keys and got to the framecounter and "delaying".

Both of us are using MacBook Air (13-inch, 2017) but he is still using Catalina while I am using Big Sur.

Could the problem be in the code itself or is the problem the keys?

jerryneedell, 2020-12-19:
An RFM69 board does not operate in LoRa mode. Since you have bypassed the version check, the code may run, but the settings for the board won't make any sense. It does not have the same register set as the RFM9x.

morrischen1, 2020-12-19:
@jerryneedell Thanks for the information! I will change board as soon as possible!

jhjulkun, 2021-02-01:
Btw, the version check fail still happens.
However, it didn't fail at least twice for me (using a M0 board, but not a Adafruit), but ONLY after a physical power cycle. Once TinyLoRa  has run once, the version check seems to fail every time (but it sends data).
Could it be a "reset", and a physical power cycle puts device in a different state?

brentru, 2021-02-01:
@jhjulkun  Could you please open another issue, this one is closed. Thank you.