adafruit/TinyLoRa #31: Lose packets LoRa 32u4 ll

camilo-17, 2019-11-25:
Hello everyone, I am working with the Lora 32u4ll of BsFrance and a gateway kernel, it works, transmits to Yeap (a lora provider) but the RSSI is very low, around -117 and my lora 32u4ll is very close of the antenna of Gateway Someone knows what happens?

I would greatly appreciate your help.

closed 2019-11-27 by camilo-17

clavisound, 2020-03-14:
Have you solved this? I have the same problem. I think the problem is after the the pull #30.

camilo-17, 2020-03-14:
Hello @clavisound see the documentation about your Gateway and look the Frequencies that it work 

Sorry if you don't understand my English its not good

clavisound, 2020-03-14:
Thank you, but I don't changed my gateway settings and I have the same results from node to node :-(

camilo-17, 2020-03-14:
Which frequency plan are you using ? And which Gateway?

clavisound, 2020-03-17:
@camilo-17 Mystery solved. For some reason #include EU863 is ignored by TinyLoRa.h Inexplicable to me. I remember it worked! 10 days of my life gone for that :-/

You are ok with #define EU863 in your sketch? Or you modified TinyLoRa.h?