adafruit/TinyLoRa #39: power off/standby api before putting arduino to sleep?

jhjulkun, 2021-02-01:
I looked but didn't see a "power off/slee" api to use before putting an arduino in deep sleep?
After looking at code, it seems like the device is put to standby after every message, but was hoping there was something that puts RFM95 in minimal power usage. I looked at the RFM95 specs, but it wasn't clear at all.

brentru, 2021-02-01:
The code puts the device into standby but lacks a power off/sleep function. I'd be open to reviewing a pull request if you'd like to add it.

jhjulkun, 2021-02-01:
Did some further testing, the drain I thought was from the radio seems to be something else, putting device in standby after each message is easier, as you don't need to begin() device next time waking from sleep.

closed 2021-02-01 by jhjulkun